Slava-Shchipunov/awg-openwrt #39: Добавить поддержку amnesiawg-1.5

Slava-Shchipunov, 2025-08-26:
Так как это форк другого репозитория, где изначально была реализована логика сборки пакетов awg для openwrt, а я лишь сделал автоматизацию сборки поверх существующего решения, то не могу обещать, что добавлю новую версию протокола. Но я попробую

@vivalaliberte отвечая на ваш вопрос, в текущей версии пакетов нет возможности конфигурировать новые параметры

vivalaliberte, 2025-09-06:
@Slava-Shchipunov
Тут - https://github.com/this-username-has-been-taken/amneziawg-openwrt _вроде как_ добавили поддержку нового протокола, но не завезли собранные .ipk файлы, их нужно компилировать самостоятельно :(

Slava-Shchipunov, 2025-09-06:
> Use amneziawg-go: it provides a newer version of the protocol (v1.5), but requires a more powerful device to run and consumes more space.

Я пробовал собирать amneziawg-go под OpenWRT, но GO пакет слишком требователен к ресурсам, на моëм AX6S он выжрал всю RAM при запуске iperf3 со скоростью больше 75 Мбит/с

Slava-Shchipunov, 2025-09-07:
> > > Use amneziawg-go: it provides a newer version of the protocol (v1.5), but requires a more powerful device to run and consumes more space.
> > 
> > 
> > Я пробовал собирать amneziawg-go под OpenWRT, но GO пакет слишком требователен к ресурсам, на моëм AX6S он выжрал всю RAM при запуске iperf3 со скоростью больше 75 Мбит/с
> 
> а из [этого](https://github.com/amnezia-vpn/amneziawg-linux-kernel-module/commits/feature/awg2/) нельзя собрать?
> 
> > feat: AWG 2.0 support
> > 
> > * global rework of build system, getting rid of patch changes
> > * support of ranged H1-H4 headers
> > * support of S3-S4 junks
> > * support of I1-I5 junk packets
> > * sync mainstream changes
> > * performance improvements

я пробовал - пакет собрался, но на роутере не заработал
но я собирал через SDK, а в репозитории, что скинули выше kmod пакет собирается из того же репозитория, но через клонирование репозитория openwrt, может это влияет. попробую сегодня также сделать, если заведётся - попробую собрать из ветки с AWG 2.0

Slava-Shchipunov, 2025-09-07:
> > я пробовал - пакет собрался, но на роутере не заработал
> 
> [вот](https://github.com/kozhini/amneziawg-openwrt) ии накодил, пакеты собираются и, вроде как, даже работают, на основе ветки авг2.0

Kmod  пакет работает? Или go пакет?

Shulyaka, 2025-09-07:
@Slava-Shchipunov Возможно ли, что Вы не сконфигурировали openwrt именно под Ваш роутер или ядро?

Slava-Shchipunov, 2025-09-07:
> [@Slava-Shchipunov](https://github.com/Slava-Shchipunov) Возможно ли, что Вы не сконфигурировали openwrt именно под Ваш роутер или ядро?

Врядли, потому что я собирал через SDK openwrt для моего роутера/версии openwrt

vivalaliberte, 2025-09-22:
@Slava-Shchipunov 

https://github.com/this-username-has-been-taken/amneziawg-openwrt/releases

v2.0.20250914
kmod-amneziawg
Added support for v2.0 protocol

Это то, что нужно?

Slava-Shchipunov, 2025-09-24:
> А эта вторая версия - она обратносоместима со старой? Если нет, то огромная просьба старый модуль (первой версии) не удалять. Это ж всю инфрастурктуру надо обновлять разом.

Разработчики заявляют обратную совместимость между версиями. Старые релизы точно никуда не денутся, но под новые релизы OpenWRT не вижу смысла собирать старую версию awg

dmitrmax, 2025-09-24:
> Разработчики заявляют обратную совместимость между версиями. Старые релизы точно никуда не денутся, но под новые релизы OpenWRT не вижу смысла собирать старую версию awg

Просто не понятно как эта обратная совместимость выглядит. Там же разные наборы параметров. И доки я чего-то не вижу на этот счет

Slava-Shchipunov, 2025-09-24:
> Просто не понятно как эта обратная совместимость выглядит. Там же разные наборы параметров. И доки я чего-то не вижу на этот счет

Ну я сейчас пробовал подключаться через новый пакет с конфигом варпа, который работает на awg1.0 - всë работает. Потом добавлял к этому конфигу I1 параметр - работает через мобильного провайдера, который блочит варп

masyanich, 2025-09-27:
> Ну я сейчас пробовал подключаться через новый пакет

а где можно взять новый пакет?)

numbereleven-a, 2025-09-29:
> Попробовал собрать, для версии [24.10.3](https://github.com/Slava-Shchipunov/awg-openwrt/releases/tag/v24.10.3) и [23.05.6](https://github.com/Slava-Shchipunov/awg-openwrt/releases/tag/v23.05.6), можно тестить
> 
> Тут в чате ITDog-a расписал [подробнее](https://t.me/itdogchat/44512/759893)

на 24ую что переходить.. эх

masyanich, 2025-09-29:
> > если вдруг кто знает где для 23.05.5 найти, отпишитесь плс
> 
> собрать 10 мин занимает - забирайте https://github.com/vortigont/amneziawg-openwrt/actions/runs/18087190724

а новый пакет для люси где живет? его бы собрать :)

masyanich, 2025-09-29:
> подойти пакет от 23.05.6,
подошел, можно вписать новый параметр, в /etc/config/network все красиво, но интерфейс не поднимается.

Mon Sep 29 10:03:02 2025 daemon.notice netifd: awg0 (12129): Unable to modify interface: Invalid argument
Mon Sep 29 10:03:07 2025 daemon.notice netifd: Interface 'awg0' is now down
Mon Sep 29 10:03:07 2025 daemon.notice netifd: Interface 'awg0' is setting up now
Mon Sep 29 10:03:08 2025 user.notice amneziawg: info: using kernel-space kmod-amneziawg for /usr/bin/awg
Mon Sep 29 10:03:08 2025 user.notice sysctl: net.ipv6.conf.awg0.disable_ipv6 = 1
Mon Sep 29 10:03:08 2025 daemon.notice netifd: awg0 (12374): Unable to modify interface: Invalid argument
Mon Sep 29 10:03:13 2025 daemon.notice netifd: Interface 'awg0' is now down
Mon Sep 29 10:03:13 2025 daemon.notice netifd: Interface 'awg0' is setting up now
Mon Sep 29 10:03:14 2025 user.notice amneziawg: info: using kernel-space kmod-amneziawg for /usr/bin/awg
Mon Sep 29 10:03:15 2025 user.notice sysctl: net.ipv6.conf.awg0.disable_ipv6 = 1
Mon Sep 29 10:03:15 2025 daemon.notice netifd: awg0 (12619): Unable to modify interface: Invalid argument

подскажите, куда копать?

Slava-Shchipunov, 2025-09-29:
честно говоря, на мой взгляд проще обновить прошивку до 24.10.3 без сохранения настроек и заново накатить пакеты

masyanich, 2025-09-29:
> честно говоря, на мой взгляд проще обновить прошивку до 24.10.3 без сохранения настроек и заново накатить пакеты

**очень, очень, очень** много всего нужного на роутере, может по доброте душевной соберете под  24.10.2?