Slava-Shchipunov/awg-openwrt #48: AWG for NanoPi R5S

BlueG1ue, 2025-09-30:
Приветствую!
Уже много времени убил на поиск и попытки сборки пакетов AWG для своего устройства Nanopi R5S.
Параметры такие:
Модель | FriendlyElec NanoPi R5S LTS
Архитектура | ARMv8 Processor 
Целевая платформа | rockchip/armv8
Версия встроенного ПО | OpenWrt 24.10.2 r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.182.04837~700a719
Версия ядра | 6.1.118

И весь нюанс заключается в том, что ядро старое. У OpenWRT 24.10.2 ядро 6.6.93. Соответственно версии под ядро 6.6.93 на мою систему не устанавливаются, а только такие и получается собрать.
Может знаете способ сборки с моим нюансом?

Slava-Shchipunov, 2025-09-30:
Привет, у вас обычная ванильная OpenWRT или какая-то сторонняя сборка?

BlueG1ue, 2025-09-30:
> Привет, у вас обычная ванильная OpenWRT или какая-то сторонняя сборка?

FriendlyWRT, основанная на OpenWrt 24.10.2, но с ядром 6.1.118